adamantine-sim/adamantine #192: Integrate heat source over time

stvdwtt, 2022-10-17:
Currently, the heat source term assumes a quasi-fixed heat source location. For large time steps, we want to integrate over the path of the heat source.

stvdwtt, 2023-01-17:
I confirmed with John Coleman that they simply sub-cycle the integration of the heat source. I think that's a reasonable first approach for us as well.

Rombur, 2023-01-19:
What we really want is to use a multi-rate method (see [here](https://sundials.readthedocs.io/en/latest/arkode/Mathematics_link.html#mristep-multirate-infinitesimal-step-methods)). We have wrappers in deal.II to use Sundials so it might be worth to use Sundials. It should be about the same amount of work than writing the code by hand and they have many multi-rate methods in case we need to switch.