adamantine-sim/adamantine #202: Lightweight postprocessing of results for the optimization framework

stvdwtt, 2022-11-15:
Motivation: To interact with the optimization framework we need access to certain quantities of the fields (e.g. temperature at a point, maximum von Mises stress).

Proposed implementation:
- Add a class that handles the postprocessing and writing the result to a text file
- Have a new type of input file that directs the postprocessing output (like the scan path, this would probably be too clunky to include in the main input file)

Initial thoughts for postprocessing options:
- Field value (x,y,z coorindates, time)
- Max field value (time)
- Min field value (time)
- Mean field value (time)

Here the field could be a primary field (e.g. temperature) or some pre-defined derived quantities (von Mises stress).

Before I start any work on this, @Rombur, does this sound reasonable? My plan would be to start with just the "field value" one and then add others once we're happy with the interface for that.

Rombur, 2022-11-15:
- Why create a new class and not extend the current `PostProcessor` class? 
- Is this something that we want to do inside adamantine or should we write a python script for VisIt? I would assume that the inputs would be a lot simpler, if you can use python. We could do something hybrid where the derived quantities are done in adamantine so we can still visualize the quantities on the entire mesh and use visit to get values at a given set of points.

stvdwtt, 2022-11-15:
Going back to the earlier plan to do the analysis in VisIt works for me. I was thinking that we'll eventually need some of this functionality in adamantine itself for the in-build optimization, but since the method there is still up in the air, waiting on it makes sense. Doing it with VisIt should be pretty quick. If/when we do need to bring some of this into adamantine, the `PostProcessor` class makes sense as a home for that functionality.

stvdwtt, 2022-12-16:
Since we went with VisIt, I'm going to close this for now. If/when we do this in adamantine, we can open a new issue more tailored to what we need.

closed 2022-12-16 by stvdwtt