adamantine-sim/adamantine #212: Zeros in the temperature introduced along MPI boundaries after adding material

stvdwtt, 2023-01-30:
Every time that material is added in an MPI run, zeros are introduced along the MPI domain boundaries. At first I assumed this was a ghost cell issue, but adding a simple `temperature.update_ghost_values()` after the call to `add_material` doesn't do anything. I'm not sure when this started happening.

@Rombur, I can keep digging, but if it's obvious to you what's wrong, let me know. Also GitHub is not letting me upload the zipped input files for a reproducer, I'll send you those directly.

stvdwtt, 2023-01-30:
Inputs:
[mpi_zeros_inputs.zip](https://github.com/adamantine-sim/adamantine/files/10540343/mpi_zeros_inputs.zip)

Temperature:
[image]

MPI domains:
[image]

Rombur, 2023-01-31:
The problem is from https://github.com/adamantine-sim/adamantine/blob/a9c99e22cf1eaad22cd22c63119276f302607ec9/source/ThermalPhysics.templates.hh#L844
If `rw_solution` contains a zero it will override the value and you will see it. 

Under the following code https://github.com/adamantine-sim/adamantine/blob/a9c99e22cf1eaad22cd22c63119276f302607ec9/source/ThermalPhysics.templates.hh#L783-L787

Can you set the value of the ghosted elements to `std::numeric_limits<double>::infinity()`

PR #217 (2023-02-01): Fix bug with ghosts zeroed out after adding material
Fixes #212 by switching from using `dealii::VectorOperation::min` to `dealii::VectorOperation::insert`, as in:
https://github.com/dealii/dealii/pull/14340

This PR also changes the 3D integration test to an MPI test to catch a similar issue in the future.

[CI](https://cloud.cees.ornl.gov/jenkins-ci/blue/organizations/jenkins/adamantine/detail/adamantine/347/pipeline)
closed 2023-02-01 by Rombur