adamantine-sim/adamantine #225: Ray intersection robustness

stvdwtt, 2023-02-22:
Under some circumstances, when the ray from experimental data is aligned with the mesh, `RayTracing::get_intersection` can erroneously give DOF index 0 as the intersection point. 

There's a note in the function that rays that cross an edge may be lost -- I think this is separate from that, although perhaps related. In the cases I saw, the ray was going through a part of the face away from the edge (and not the exact middle of the face either).

When I have a chance, I'll post a reproducer.

Rombur, 2023-02-28:
Looking at the code, there are several problem with the current function:
 1. This code below needs to use some kind of tolerance https://github.com/adamantine-sim/adamantine/blob/1a42819437d702713b351cddbeb1287e2b232dfe/source/experimental_data.cc#L592-L608
 2. The tolerance used here https://github.com/adamantine-sim/adamantine/blob/1a42819437d702713b351cddbeb1287e2b232dfe/source/experimental_data.cc#L567-L570
 should be a relative tolerance not an absolute one.

Rombur, 2025-04-23:
Let's close this issue for now. This code has changed significantly. We can reopen it if the problem reappears.

closed 2025-04-23 by Rombur