adamantine-sim/adamantine #247: Issue with material deposition in parallel

stvdwtt, 2023-10-04:
In series material deposition happens as expected. However in parallel some of the mesh is not activated as expected.

Serial:
[image]

2 MPI processes:
[image]

4 MPI processes:
[image]

h/t Ashley Gannon

Rombur, 2023-10-04:
Can you post the partitioning? It's the `subdomain_id` plot

stvdwtt, 2023-10-05:
The material is only being deposited in the subdomain for MPI process 0:
[image]

Rombur, 2023-10-05:
Can  you share your input files. I tried to reproduce the problem by modifying one of our test but it works as expected.

stvdwtt, 2023-10-05:
Here you go:
[inputs.zip](https://github.com/adamantine-sim/adamantine/files/12819435/inputs.zip)

Rombur, 2023-10-05:
The code crashes for me in parallel when using your input files. I assume that you are running in release mode and that's why the code keeps running

Rombur, 2023-10-05:
I have a fix. I'll open a PR soon.

PR #248 (2023-10-05): Fix a bug when depositin material in parallel
This PR fixes #247 The way the code was working was the following: we compute all the cells that are activated as part of the material deposition and we associate a time stamp them. Then we need to add material, we search for the cells that need to be activated given their time stamp. When the mesh is refine, we recompute the new cells to activate. The issue is that adding material triggers a mesh repartitioning and the cells that needed to be activated may have moved to a different processor. In debug mode this causes the code to crash because we  [... cut]
closed 2023-10-05 by stvdwtt