adamantine-sim/adamantine #271: Automate CFL time step calculation

stvdwtt, 2024-04-16:
It should be straightforward to compute the global minimum time step for a simulation, assuming that we know the max thermal conductivity, the minimum coarse mesh element size, and the maximum number of refinements. This would allow us to make it so the user doesn't need to specifically pick the time step size, which would make things easier. We could add a user-specified safety factor so they can decrease the time step if desired.

Later on we could do a more sophisticated version of this where we adapt based on the layer that we're on, but that's out of scope for now.

Rombur, 2024-05-06:
This should be high on our priority list. Right now it's really difficult for users to pick a good time step a priori. It's really cumbersome to use adamantine with Slicer 2

stvdwtt, 2024-09-08:
Note to myself here, use dealii.GridTools::minimal_cell_diameter().

Rombur, 2025-09-26:
Relevant paper https://www.mdpi.com/1099-4300/27/10/994

AshGannon, 2026-11-03:
Comment/note to self - implementation should inform the `time_steps_between_output` input or we should change this input to be in seconds. 

Use case: lining up in-situ data with simulation outputs for calibration or assimilation.

Rombur, 2026-11-03:
The `time_steps_between_output`  is a bad input. I want to replace it. I need to think more about the replacement but right know I want the user to give a formula that we evaluate and which tell us when we need to write an output. For instance, if you want an output after 0.5 s, 10.5 s, and so on. You would write `10*t + 0.5`. We can use muparser to parse the formula so that we can even have `if` conditions in the formula.