adamantine-sim/adamantine #313: can the code simulate the temperature field at part-level?

mikejwg, 2024-09-03:
Dear authors,

I am interested in simulating the temperature fields at part-level (for instance, 100mm SLMed part) and export the temperature information at the varied locations (through defining specified Node sets) to Celluar Automa (CA) code to predict the microstructure evolution during SLM.  I used Abaqus, but if the temperature field is required, only process-level fine mesh size and time increment must be defined in the transient heat transfer FE simulation.  Could adamantine simulate the temperature fields at part-level?
Thanks very much.

Rombur, 2024-09-03:
Hello,

Yes, adamantine can do that. However, we currently only output the temperature in `vtu` files for visualization with VisIt or Paraview. We do not write a list of points and temperatures which I assume is what you want.

mikejwg, 2024-09-04:
Thank you for your reply. I believe that ParaView can output temperature information for a specified region. However, I am seeking documentation on how to use Adamantine specifically for simulating at the part-level, as the provided help files are not sufficient to address our current needs. Could you please guide me on where I can download or access more comprehensive documentation for Adamantine? This would greatly assist us in tackling the issue.

stvdwtt, 2024-09-04:
Hi @mikejwg, actually our use of adamantine is focused on simulations at the part level, so that isn't an issue. I know our documentation is a bit sparse at the moment -- we're working on a tutorial that walks through a few example problems.

Right now the best example to run is the Hourglass_AOP test. You can find the input files in `tests/data/`. If you grab the three `Hourglass_AOP*` files, you'll see an example of a part scale simulation. That uses a castellated (i.e. voxelized) mesh, but we can import body-fitted meshes too.

Finally, I'm not sure what CA code you're using, but you may be interested in ExaCA, which our colleagues develop:
https://github.com/LLNL/ExaCA

We don't have a direct interface to ExaCA yet, but that's part of our roadmap.