adamantine-sim/adamantine #314: Seg fault during covariance matrix construction for data assimilation only for particular numbers of MPI tasks

stvdwtt, 2024-09-08:
Summary: An IMTS gear simulation crashes after data assimilation only for the case where 27 MPI processes are used.

Test case:
https://github.com/adamantine-sim/demonstration-cases/tree/main/IMTS_Parts/GEAR-OP04

To run this, copy the input files out of `simulation_template` into its parent `GEAR-OP04` directory and then run adamantine.

We get a seg fault at this line:
https://github.com/adamantine-sim/adamantine/blob/587a4ead79730ea7b30829346766f7e8bc781598/source/DataAssimilator.cc#L581

with an index that is greater than the number of degrees of freedom.

It's unclear whether the issue is in `indices_ranks` or if `j` has an invalid value.

@Rombur, could you take a look at this?

Rombur, 2024-09-09:
Ughh it passed on my machine. My guess is that due to numerical error, ArborX is giving us results we cannot deal with. Have you tried running in debug mode?

stvdwtt, 2024-09-09:
Yeah, debug mode is how I localized it. It fails with 27 ranks on the SCOPS-foundry-2 computer at the MDF