adamantine-sim/adamantine #316: Adamantine hangs during ArborX call for the IMTS mold mesh

stvdwtt, 2024-09-11:
For the particular input files below, rank 0 of adamantine hangs in `DataAssimilator::update_covariance_sparsity_pattern`.

It happens during this ArborX call:
https://github.com/adamantine-sim/adamantine/blob/36131aaaa13433d2d5b6cb4b998e6c5d8f8862f8/source/DataAssimilator.cc#L541

The backtrace is rather long:
```#0  ArborX::Details::LegacyCallbackWrapper<ArborX::Details::DefaultCallbackWithRank>::operator()<ArborX::Intersects<ArborX::Sphere>, ArborX::Box, ArborX::Details::InsertGenerator<ArborX::Details::FirstPassNoBufferOptimizationTag, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, true>, ArborX::Details::LegacyCallbackWrapper<ArborX::Details::DefaultCallbackWithRank>, Kokkos::View<Kokkos::pair<int, int>*, Kokkos::HostSpace>, Kokkos::View<int*, Kokkos::HostSpace>, ArborX::Details::PermutedData<Kokkos::View<int*, Kokkos::HostSpace>, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, false> >::operator()<ArborX::Details::PairIndexVolume<ArborX::Box> >(ArborX::PredicateWithAttachment<ArborX::Intersects<ArborX::Sphere>, int> const&, ArborX::Details::PairIndexVolume<ArborX::Box> const&) const::{lambda(Kokkos::pair<int, int> const&)#1}>(ArborX::Intersects<ArborX::Sphere> const&, ArborX::Details::PairIndexVolume<ArborX::Box> const&, ArborX::Details::InsertGenerator<ArborX::Details::FirstPassNoBufferOptimizationTag, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, true>, ArborX::Details::LegacyCallbackWrapper<ArborX::Details::DefaultCallbackWithRank>, Kokkos::View<Kokkos::pair<int, int>*, Kokkos::HostSpace>, Kokkos::View<int*, Kokkos::HostSpace>, ArborX::Details::PermutedData<Kokkos::View<int*, Kokkos::HostSpace>, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, false> >::operator()<ArborX::Details::PairIndexVolume<ArborX::Box> >(ArborX::PredicateWithAttachment<ArborX::Intersects<ArborX::Sphere>, int> const&, ArborX::Details::PairIndexVolume<ArborX::Box> const&) const::{lambda(Kokkos::pair<int, int> const&)#1} const&) const (this=0x7ffd1d31f118, predicate=..., value=..., out=...) at /opt/arborx/include/ArborX/details/ArborX_DetailsLegacy.hpp:75
#1  0x000055b51eddfabe in ArborX::Details::InsertGenerator<ArborX::Details::FirstPassNoBufferOptimizationTag, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, true>, ArborX::Details::LegacyCallbackWrapper<ArborX::Details::DefaultCallbackWithRank>, Kokkos::View<Kokkos::pair<int, int>*, Kokkos::HostSpace>, Kokkos::View<int*, Kokkos::HostSpace>, ArborX::Details::PermutedData<Kokkos::View<int*, Kokkos::HostSpace>, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, false> >::operator()<ArborX::Details::PairIndexVolume<ArborX::Box> > (
    this=0x7ffd1d31f118, predicate=..., value=...) at /opt/arborx/include/ArborX/details/ArborX_DetailsCrsGraphWrapperImpl.hpp:92
#2  0x000055b51ee3077a in ArborX::Details::invoke_callback_and_check_early_exit<ArborX::Details::InsertGenerator<ArborX::Details::FirstPassNoBufferOptimizationTag, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, true>, ArborX::Details::LegacyCallbackWrapper<ArborX::Details::DefaultCallbackWithRank>, Kokkos::View<Kokkos::pair<int, int>*, Kokkos::HostSpace>, Kokkos::View<int*, Kokkos::HostSpace>, ArborX::Details::PermutedData<Kokkos::View<int*, Kokkos::HostSpace>, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, false> > const&, ArborX::PredicateWithAttachment<ArborX::Intersects<ArborX::Sphere>, int> const&, ArborX::Details::PairIndexVolume<ArborX::Box> const&> (callback=..., predicate=..., primitive=...)
    at /opt/arborx/include/ArborX/details/ArborX_Callbacks.hpp:166
#3  0x000055b51ee2e6a7 in ArborX::Details::TreeTraversal<ArborX::BasicBoundingVolumeHierarchy<Kokkos::HostSpace, ArborX::Details::PairIndexVolume<ArborX::Box>, ArborX::Details::DefaultIndexableGetter, ArborX::Box>, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, true>, ArborX::Details::InsertGenerator<ArborX::Details::FirstPassNoBufferOptimizationTag, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, true>, ArborX::Details::LegacyCallbackWrapper<ArborX::Details::DefaultCallbackWithRank>, Kokkos::View<Kokkos::pair<int, int>*, Kokkos::HostSpace>, Kokkos::View<int*, Kokkos::HostSpace>, --Type <RET> for more, q to quit, c to continue without paging--c
ArborX::Details::PermutedData<Kokkos::View<int*, Kokkos::HostSpace>, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, false> >, ArborX::Details::SpatialPredicateTag>::operator()<ArborX::PredicateWithAttachment<ArborX::Intersects<ArborX::Sphere>, int> > (this=0x7ffd1d31f090, predicate=...) at /opt/arborx/include/ArborX/details/ArborX_DetailsTreeTraversal.hpp:110
#4  0x000055b51ee2b0a4 in ArborX::Details::TreeTraversal<ArborX::BasicBoundingVolumeHierarchy<Kokkos::HostSpace, ArborX::Details::PairIndexVolume<ArborX::Box>, ArborX::Details::DefaultIndexableGetter, ArborX::Box>, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, true>, ArborX::Details::InsertGenerator<ArborX::Details::FirstPassNoBufferOptimizationTag, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, true>, ArborX::Details::LegacyCallbackWrapper<ArborX::Details::DefaultCallbackWithRank>, Kokkos::View<Kokkos::pair<int, int>*, Kokkos::HostSpace>, Kokkos::View<int*, Kokkos::HostSpace>, ArborX::Details::PermutedData<Kokkos::View<int*, Kokkos::HostSpace>, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, false> >, ArborX::Details::SpatialPredicateTag>::operator() (this=0x7ffd1d31f090, queryIndex=6905) at /opt/arborx/include/ArborX/details/ArborX_DetailsTreeTraversal.hpp:98
#5  0x000055b51ee24933 in Kokkos::Impl::ParallelFor<ArborX::Details::TreeTraversal<ArborX::BasicBoundingVolumeHierarchy<Kokkos::HostSpace, ArborX::Details::PairIndexVolume<ArborX::Box>, ArborX::Details::DefaultIndexableGetter, ArborX::Box>, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, true>, ArborX::Details::InsertGenerator<ArborX::Details::FirstPassNoBufferOptimizationTag, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, true>, ArborX::Details::LegacyCallbackWrapper<ArborX::Details::DefaultCallbackWithRank>, Kokkos::View<Kokkos::pair<int, int>*, Kokkos::HostSpace>, Kokkos::View<int*, Kokkos::HostSpace>, ArborX::Details::PermutedData<Kokkos::View<int*, Kokkos::HostSpace>, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, false> >, ArborX::Details::SpatialPredicateTag>, Kokkos::RangePolicy<Kokkos::Serial, ArborX::Details::TreeTraversal<ArborX::BasicBoundingVolumeHierarchy<Kokkos::HostSpace, ArborX::Details::PairIndexVolume<ArborX::Box>, ArborX::Details::DefaultIndexableGetter, ArborX::Box>, ArborX::Details::PermutedData<Kokkos::View<ArborX::Intersects<ArborX::Sphere>*, Kokkos::Device<Kokkos::Serial, Kokkos::HostSpace> >, Kokkos::View<unsigned int*, Kokkos::Device<Kokkos::Serial, Kokk [... cut ...] ine/application/adamantine.hh:2040
#22 0x000055b51e88c609 in main (argc=3, argv=0x7ffd1d32f068) at /home/adamantine/application/adamantine.cc:709```

Inputs:
[failing_example.zip](https://github.com/user-attachments/files/16965370/failing_example.zip)

For the three ensemble members in his failing test case, I've seen it hang for 1, 3, and 9 MPI ranks. Fortunately, that means it can be debugged with serial tests.

I wonder if this and #314 have the same root cause.

@Rombur, can you take a look?

Rombur, 2024-09-11:
@stvdwtt the `build_time_log.txt` and the `.csv` files are missing

stvdwtt, 2024-09-11:
Sorry about that, here they are:
[observations.zip](https://github.com/user-attachments/files/16967738/observations.zip)

stvdwtt, 2024-09-24:
Closing since this isn't an ArborX issue, it's a user-issue.

closed 2024-09-24 by stvdwtt