adamantine-sim/adamantine #319: joss-paper: Minor improvements of software paper

ericneiva, 2024-09-20:
Dear @Rombur,

First of all, I would like to commend you for a very clear and well-structured software paper.

Please, find below some minor suggestions for improvement (some of them have already been pointed out by other reviewers):

* [ ] _Line 7_ Please cite Trilinos as suggested by the project authors. See [Citing Trilinos](https://trilinos.github.io/cite.html).
* [ ] _Line 9_ Direct -> Directed
* [ ] _Line 11_ For completeness, I suggest to also mention the other simulated physics (mechanics and thermomechanics).
* [ ] _Line 23_ obtained -> obtain
* [ ] _Line 31_ Duplicate definition of IR acronym (see _Line 16_)
* [ ] _Line 62_ fosr -> for
* [ ] _Line 63_ assosciated -> associated
* [ ] _Line 70_ stocastic -> stochastic
* [ ] _Line 71_ Duplicate definition of EnKF acronym
* [ ] _Line 140_ Review DOI
* [ ] _Line 145_ Review DOI

See https://github.com/openjournals/joss-reviews/issues/7017

Rombur, 2024-09-24:
Everything on the list is done. @ericneiva I'll let you close the issue if you are happy with the result.

ericneiva, 2024-09-24:
Nice! Thanks a lot @Rombur for taking care of all the comments!

closed 2024-09-24 by ericneiva