adamantine-sim/adamantine #320: joss-paper: Improvements in README.md

ericneiva, 2024-09-20:
Dear @Rombur,

Please, find below some suggestions to improve the documentation of the README.md file:

Also note that some of the changes proposed here should be mirrored in the documentation website.

* [ ] _Adamantine logo:_ It cannot be seen in dark mode
* [ ] _Statement of need:_ I suggest to elaborate a little more at the beginning about what problems the software can solve and who the target audience is, as in the web documentation.

#### Installation

* [ ] [p4est](https://www.p4est.org/) should be in small caps and added as a dependency in the bullet list (via deal.ii)
* [ ] Perhaps it's worth adding the optional dependencies in the bullet list for clarity and completeness, that is,
  * _Optional_ SEACAS
  * _Optional_ Caliper

#### Docker

* [ ] Please, extend the information to match the one of the web documentation

#### Input file

* [ ] Add or link to an example _input file_ for the purpose of illustration
* [ ] For the sake of clarity, maybe it will help to format the options in a table (name, values, units, required, comments)
* [ ] The input file description is very informative and exhaustive, but as it is so detailed, maybe the README.md will be more readable, if you put it into a separate markdown file and then link to it
* [ ] Several parameters lack the units. Here, a non-exhaustive list:
  * [ ] *material_height*
  * [ ] *deposition_length*
  * [ ] *deposition_width*
  * [ ] *deposition_lead_time*
  * [ ] *deposition_time*
  * [ ] *initial_temperature*
  * [ ] *new_material_temperature*
  * [ ] ...

### Scan path

  * [ ] Link to examples of the _segment_ and _event_ formats

### Material deposition

  * [ ] Remove `\` from `deposition\_length` and the rest of parameters.
  * [ ] Please link to an example file for file-based deposition.

### Example usage

  * [ ] Please, link to the examples in the web documentation to avoid searching for them
  * [ ] _In the web documentation_ To facilitate running the examples, provide the execution command for each sub-example of the Demo316 and HourGlass folders

See https://github.com/openjournals/joss-reviews/issues/7017

Rombur, 2024-09-24:
@ericneiva I've done all the changes but two for the `input` section. We agree that the `input` section in the README can be improved but we would like more time to find the best way to present the information. If it's ok with you, we will keep the section as-if for now.

ericneiva, 2024-09-25:
Hi, @Rombur, sounds good to me. Thanks for applying all the other changes.

I will check off the remaining items in https://github.com/openjournals/joss-reviews/issues/7017 meaning my review is complete. 

Do you want to keep this issue open to not forget about the remaining comments or you'd rather close it?

Rombur, 2024-09-25:
I would prefer if you close the issue since this is issue is for the paper.

closed 2024-09-26 by ericneiva