adamantine-sim/adamantine #324: vtk

a1227659796, 2024-09-28:
Hello author, may I ask how the vtk file is generated? What software is used to generate it? Are there any requirements for the software version? Also, what types of meshes does it support? Can it also simulate irregular shapes? I'm looking forward to your answer, thank you.

Rombur, 2024-09-30:
Our code is based on [deal.II](https://dealii.org/) and they have a function a function (see [here](https://dealii.org/current/doxygen/deal.II/namespaceDataOutBase.html#afcc7a28f382c0bc409bae1aa1aff7591)) that creates the vtk file for us. deal.II is a general finite element library not a specialized library to create vtk file. Since the code is open source, you can see how the function is implemented. I know the code works for triangles, quadrilaterals, tetrahedrons, hexahedrons,  pyramids, and wedges.

a1227659796, 2024-09-30:
Thank you for your reply. Since I use Abaqus frequently, I noticed that the mesh input section also supports Abaqus format files. Are there any specific format requirements for the .inp mesh file? The model I generated using the .inp mesh file doesn't show any results in the PVD file. The specific runtime log is as follows. What could be the issue? I look forward to your response.
# mpirun -n 2 ./adamantine --input-file=test1.info
Starting non-ensemble simulation
n_time_step: 1 time: 0 n_dofs: 2002
n_dofs after cell activation: 2002
10% completed
20% completed
30% completed
40% completed
50% completed
60% completed
70% completed
80% completed
90% completed
100% completed
Simulation done

Rombur, 2024-09-30:
Can you share the `test1.info` file. It's hard to tell without seeing the input file. However since the code runs to completion, I don't think that the issue is with the mesh. There is an input parameter `time_steps_between_output` that controls  how often the `vtk` files are written. Make sure that this number is reasonable. In some of our tests, we don't want to create output files and we set that number to a very high value.

Also I may have misunderstood your previous questions. While there is little restriction on the `vtk` output, adamantine only supports conformal meshes with hexahedra.

a1227659796, 2024-10-07:
[zip.zip](https://github.com/user-attachments/files/17279526/zip.zip)
This is my executable file, but the result that came out looks like the one above.

Rombur, 2024-10-08:
Thanks, I'll take a look as soon as possible.

Rombur, 2024-10-10:
The problem is in the scan path. In your `scan.txt`, a lot of your segments use the spot mode for zero second and zero power. This does nothing but "move" instantly the turned off heat source. The line segments have their end point identical to their starting point and so nothing happens. Since there is no active source, nothing happens and you just see the initial conditions.

a1227659796, 2024-10-11:
Thank you very much. If I have a gcode file, how can I quickly obtain the scan path in the specialized adamantine format? Manually creating the path would take a long time.

a1227659796, 2024-10-11:
Can this program simulate the SLM process? When I set the laser to 200W and the moving speed to 1m/s, the resulting model temperature is over 300 degrees, and the simulated temperature path looks discontinuous, more like individual points.
[image]

stvdwtt, 2024-10-11:
What's your time step? The heat source is only calculated at the time steps, so if your velocity is high and your time step is large, you might see that behavior. To properly resolve the source, you should make sure that the laser is moving less than an element's width per time step.

a1227659796, 2024-10-12:
Thank you for your explanation. The issue mentioned earlier has disappeared. However, I want to simulate the SLM printing process, and I set the laser power to 300W, but the resulting temperature field is still very low, only around 300 degrees. Is this software only capable of simulating electron beam additive manufacturing? Does it support SLM? Do I need to increase the maximum power to something like 2000W or more?

Rombur, 2024-10-14:
The temperature in the part depends on source parameters and the material properties. These depend on the experiment that you are trying to simulate. I don't know what values make sense for your particular problem.