adamantine-sim/adamantine #331: Nix Build

wd15, 2024-10-28:
I'm trying to generate a Nix build for Adamantine. 

https://github.com/wd15/adamantine/blob/nix/flake.nix

The dependencies seem to build without an issue, but Adamantine is failing with

```
[ 55%] Building CXX object source/CMakeFiles/Adamantine.dir/ThermalOperatorDeviceInstSDev.cc.o                                                                                                 
In file included from include/boost/filesystem/exception.hpp:18,                                                                                                                               
                 from include/boost/filesystem/operations.hpp:25,                                                                                                                              
                 from /nix/store/33gxi5bk8mkcynrg712ikikx70mw4vzq-deal_II-9.5.2/include/deal.II/bundled/boost/filesystem.hpp:17,                                                               
                 from /build/adamantine/source/ScanPath.hh:15,                                                                                                                                 
                 from /build/adamantine/source/ScanPath.cc:8:                                                                                                                                  
/nix/store/33gxi5bk8mkcynrg712ikikx70mw4vzq-deal_II-9.5.2/include/deal.II/bundled/boost/system/error_code.hpp:932:10: fatal error: boost/system/detail/system_category_posix.hpp: No such file>
  932 | #include <boost/system/detail/system_category_posix.hpp>                                                                                                                               
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                               
compilation terminated.                                   
```

I'm basing the build on the [Dockerfile](https://github.com/adamantine-sim/adamantine/blob/56fe8b15fe2bb413129a03c39b9dfa136639873e/ci/Dockerfile#L4) with consistent versions. Any pointers that might help on the above failure? Thanks.

masterleinad, 2024-10-28:
The `Dockerfile` has an external boost installation. Thus, it looks suspicious to see the bundled boost in your error message.

Rombur, 2024-10-28:
@wd15 both adamantine and deal.II use Boost. deal.II bundles a strip-down version of Boost that is used if Boost is not found. Like @masterleinad said, it looks like the bundle version has been used. The issue is that adamantine uses functions that are not part of the bundled Boost. You need to make sure that both adamantine and deal.II uses the same complete version of Boost.

wd15, 2024-11-07:
Thanks for following up. When an external boost is included with deal.II as suggested above it gives the following error. I think this is close to working. Nix should be starting with a clean temp area for every build so there shouldn't be left over build files as suggested by the error message.

```
                                                                                                                                                                                              
-- Found Boost: /nix/store/lp4p7nlgnqjj0s62nb79lh0xkazgpp4d-boost-1.83.0-dev/lib/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.70.0") found co>
-- Found Adiak: /nix/store/fnw64v6f8s6lzzflg36v8y3ljy829d7f-adiak-0.4.0                                                                                                                        
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD                                                                                                                                                     
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success                                                                                                                                           
-- Found Threads: TRUE                                                                                                                                                                         
-- Found Caliper: /nix/store/1xidzhm3yzhsimd4q54wzrslha5471ci-caliper-2.10.0                                                                                                                   
-- Build type: Release                                                                                                                                                                         
-- Configuring done (2.1s)                                                                                                                                                                     
CMake Error in application/CMakeLists.txt:                                                                                                                                                     
  Imported target "dealii::dealii_release" includes non-existent path                                                                                                                          
                                                                                                                                                                                               
    "/nix/store/ldwlpwk95qmvfr6v3akjyb4dhcprdcab-deal_II-9.5.2/include/deal.II/bundled"                                                                                                        
                                                                                                                                                                                               
  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:                                                                                                                             
                                                                                                                                                                                               
  * The path was deleted, renamed, or moved to another location.                                                                                                                               
                                                                                                                                                                                               
  * An install or uninstall procedure did not complete successfully.                                                                                                                           
                                                                                                                                                                                               
  * The installation package was faulty and references files it does not                                                                                                                       
  provide.

CMake Error in source/CMakeLists.txt:
  Imported target "dealii::dealii_release" includes non-existent path

    "/nix/store/ldwlpwk95qmvfr6v3akjyb4dhcprdcab-deal_II-9.5.2/include/deal.II/bundled"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```

For reference, this is the [flake.nix](https://github.com/wd15/adamantine/blob/a0793f983c70d371d0455252937670db8e683312/flake.nix)

wd15, 2024-11-07:
Thanks for your help. Built and runs on of the test examples. Shall I submit as a PR? I'll clean it up some first. A small test case can be included in the build so which one would be the most lightweight to include from the example files?

Rombur, 2024-11-08:
> Shall I submit as a PR?

Yes, please

> A small test case can be included in the build so which one would be the most lightweight to include from the example files?

This one `test_integration_2d` would be good. It runs in a few seconds.  You can also run it with `ctest -R integration_2d`.

closed 2025-03-11 by wd15