adamantine-sim/adamantine #338: Radiative and convective BCs cool the part too quickly

stvdwtt, 2024-11-12:
We should do some more verification of the convective and radiative boundary conditions, they seem too strong. E.g. with just radiative conditions a part will cool from 1000K to 300K in 30 s. Similarly to get reasonable answers it looks like the convective coefficient needs to be on the order of 0.1, when references online suggest that 1-10 W/(m^2*K) is normal for free convection.

Ideally we'd find a good analytic test case, but we can also compare to e.g. AdditiveFOAM.

Rombur, 2024-12-05:
>  just to confirm, is the one in ThermalPhysics only used for GPU calculations and will be removed once the source calculation on the GPU is complete?

Yes, this only applies to the GPU calculations. See: 
https://github.com/adamantine-sim/adamantine/blob/d93ead353d087af286d6c7784abea0b70cbf5a94/source/ThermalPhysics.templates.hh#L87-L91

We don't compute the boundary conditions in `ThermalOperatorDevice`.

Rombur, 2025-04-23:
@AshGannon I think your working on this. If you are not feel free to unassign yourself.

AshGannon, 2025-04-23:
On it! Thank you for adding me to this issue.

AshGannon, 2025-07-01:
@Rombur you're right! Great catch.

closed 2025-07-02 by Rombur