adamantine-sim/adamantine #348: Confusion adding a powder layer

wd15, 2024-12-06:
I am having some difficulties adding a powder layer. The files in [this Gist]( https://gist.github.com/wd15/e9cd6ce1ebbfc2f987ed841c074515f6) model a block with a single heat source and a 2x2x2 mesh. Half the block should be solid and half the block should be powder (at least from my understanding). I've set the geometry section up as follows to achieve a half powder, half solid scenario for debugging purposes.

```
geometry
{
  import_mesh false ; Use built-in mesh generator
  dim 3 ; dimension of the domain
  length 1.0e-3 ; [m]
  height 1.0e-3 ; [m] In 3D, the third parameters is width
  width  1.0e-3
  length_divisions 2 ; Number of cell layers in the length direction
  height_divisions 2 ; Number of cell layers in the height direction
  width_divisions  2
  use_powder true
  powder_layer 0.5e-3
  material_height 1.0e-3
}
```

When the simulation is run for just a few steps, the output is as follows (using meshio to view the data values for the initial condition).

```python
>>> import meshio
>>> mesh = meshio.read("output.0.0.vtu")
>>> mesh.point_data['powder']
array([0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.], dtype=float32)
>>> mesh.point_data['solid']
array([1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1.,
       1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.], dtype=float32)
```

And after the first time step

```
>>> import meshio
>>> mesh = meshio.read("output.1.0.vtu")
>>> mesh.point_data['powder']
array([0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.], dtype=float32)
>>> mesh.point_data['solid']
array([1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1.,
       1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1.,
       1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1.,
       1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1., 1.], dtype=float32)
```

Is this behavior correct? Firstly, should the powder be labelled one where the solid is zero and should the solid region change after only a single step in these output files? 

I did try stepping through this with the debugger, specifically in the `assign_material_state` function in the `Geometry.cc` file. It seems to work correctly at that point assigning equal cells to the powder and solid state (entering each branch 4 times for the 8 cells in the domain).

Rombur, 2024-12-06:
There seems to be a problem in the initial assignment. For each cell the sum of `powder`, `solid`, and `liquid` should be equal to one. Since the `powder` does not seem to be assigned correctly, after one time step everything is set to `solid` so that the sum equals one. I'll take a look.

Rombur, 2024-12-06:
I found the bug. The number of material states, between 1 and 3, is a template parameter. This allows us to efficiently loop over the material states because the loop bound is known at compile time. We read the input file, detect the number of material states, and then call the right templated class. The issue is that we were off by one when computing the number of material states. We never realized it because:
1. we only do DED but our input files often have powder data that we forget to remove
2. our integration tests don't call the executable directly and so one single file (`adamantine.cc`) is never tested. I let you guess where the bug was.

wd15, 2024-12-12:
I just tested with the latest version and works for me. Thanks.