adamantine-sim/adamantine #363: Explanation of data assimilation output files

wd15, 2025-02-05:
Regarding the [Bare plate L da augmented example](https://adamantine-sim.github.io/adamantine/doc/examples/bare_plate_l#bare-plate-l-da-augmented). I'll like to understand how to access or output the updated or improved absorption coefficient. I'm assuming the value stated in the data output files like `output_m0_data.txt` are the values of the absorption coefficient randomly chosen at the beginning of each ensemble simulation, not the updates (updated via DA) that are calculated during the simulation. Does that make sense?

stvdwtt, 2025-02-06:
You're right that the output files like `output_m0_data.txt` are only for the initial values. Currently the only place we output the updated values from DA is what we print to standard output. But it would be a good idea to write those somewhere to record how the estimate varies over the course of the run.

wd15, 2025-02-06:
Thanks for the prompt reply! I didn't notice the standard output. I was getting frustrated as I was grepping for code in the `source` directory and then I realized that some of the code is in the `application` directory. I can mess with this area of the code,

https://github.com/adamantine-sim/adamantine/blob/master/application/adamantine.hh#L2157

and dump the output to files.

Rombur, 2025-02-07:
>  I was getting frustrated as I was grepping for code in the source directory and then I realized that some of the code is in the application directory.

Yeah I don't know if it was a good idea to separate the code in different directories. The reason I did that is because in `source` you have the `Adamantine` library and in `application`, you have the `adamantine` program built on the top of the library. We build a library because it's easier to test a library than an executable.

wd15, 2025-02-07:
> Yeah I don't know if it was a good idea to separate the code in different directories. The reason I did that is because in `source` you have the `Adamantine` library and in `application`, you have [...]

No worries! Design decisions are difficult. It's a fantastic code that will only get better with more users.

closed 2025-03-11 by wd15