adamantine-sim/adamantine #376: GPU Issue

wd15, 2025-03-11:
The attached input file works with regular MPI build Adamantine.

Rombur, 2025-03-11:
> Basically, is this behavior expected?

Yes, it is expected. We should check that you run on the CPU when `mechanical` is true. Just to be sure, it only fails when you set `memory_space` to `default` right? In your input file you have `memory_space host` which should always work. It is OK to build adamantine with CUDA support and not use it.

wd15, 2025-03-12:
Here is a table of results for my recent GPU build. 

| Build                                                    | mechanical | memory_space | wall time (s) |
| -------------                                            | -------------    | --------------------  | ---------------  |
| GPU configured (local non-nix build)  | false            | device               | 12083          |
| GPU configured (local non-nix build)  | false            | host                  | 1763            |
| GPU configured (local non-nix build)  | true            | host                   | seg-fault       |
| GPU configured (local non-nix build)  | true            | device                | seg-fault       |
| Nix build (not-GPU)                             | true             | host                   | 2209            |

It seg-faults for both `memory_space` options with mechanical. That might indicate an issue with the build. The Nix build works just fine for this problem. Also, when running on the GPU things are really slow. The problem has about 11K cells.

Rombur, 2025-03-12:
The host build should always work. I'll try to reproduce the issue.

> Also, when running on the GPU things are really slow. 

The big issue with the GPU is that the heat source is computed on the host at every time step and then moved to the device. We started to work on fixing this issue last year https://github.com/adamantine-sim/adamantine/pull/282 but we ran out of funding before we had time to finish it.

wd15, 2025-03-12:
Thanks for the reply. Good to know the issue regarding the heat source.

Rombur, 2025-03-14:
@wd15 Let me hijack this issue since you've put some wall times. Part of deal.II can use vector instructions when you compile with `-march=native`. We cannot use that option in Docker because the user's machine may not support the same instructions. On my local build, I see a 10% speedup for the operator evaluation and I have ideas on to improve that. I wonder if it would make sense to compile deal.II and adamantine with `-march=native` for the Nix build? I don't know if people cross-compile with Nix or use it to create Docker image, in which case the binary may not run.

wd15, 2025-03-14:
> [@wd15](https://github.com/wd15) Let me hijack this issue since you've put some wall times. Part of deal.II can use vector instructions when you compile with `-march=native`. We cannot use that [...]

It can certainly be an option to have. I'll try it out. Should the option be treated like a cmake flag and added [here](https://github.com/adamantine-sim/adamantine/blob/master/nix/adamantine/common.nix#L40) for example?

> I don't know if people cross-compile with Nix or use it to create Docker image, in which case the binary may not run.

Nix can be used for cross-compilation and creating Docker images. Let's see if I can get it working with the `-march=native` flag.

Rombur, 2025-03-17:
> Should the option be treated like a cmake flag and added [here](https://github.com/adamantine-sim/adamantine/blob/master/nix/adamantine/common.nix#L40) for example?

Yes, exactly. You would need to add `-DCMAKE_CXX_FLAGS="-march=native"` both for the deal.II and the adamantine builds

wd15, 2025-03-20:
Just a note that I don't see any speed up with `-march=native` on some speed tests. It did build though. That's with building using `-march=native` on a cluster head node and then deploying to the cluster.

Rombur, 2025-03-21:
That's pretty disappointing. Thanks for trying it.