adamantine-sim/adamantine #380: Error Running Hourglass Example from Nix Installation

dpagendam, 2025-04-08:
Hi, I hope this is the right place to post about the error I am encountering - I am sure there is a simple solution to it but I am not very familiar with using Nix.

I installed Adamantine using the instrutions for Nix at the bottom of this page:
https://adamantine-sim.github.io/adamantine/doc/install.html

I am trying to run the Hourglass example that is provided here:
https://adamantine-sim.github.io/adamantine/doc/examples/hour_glass.html

I begin by running:
nix develop github:adamantine-sim/adamantine --experimental-features 'nix-command flakes'

I then get some warning messages and an error.  These are:
warning: ignoring untrusted substituter 'https://mdfbaam.cachix.org', you are not a trusted user.
Run `man nix.conf` for more information on the `substituters` configuration option.
warning: ignoring the client-specified setting 'trusted-public-keys', because it is a restricted setting and you are not a trusted user
error: creating temporary file '/tmp/nix-shell.5llXsN/nix-shell.b6vR4V/nix-shell.oeIUEE': No such file or directory

If I instead run the above command with sudo, as:
sudo nix develop github:adamantine-sim/adamantine --experimental-features 'nix-command flakes'

I get a "root#" prompt.  If I then try to run the example with:
mpirun -np 1 ./adamantine -i HourGlass_AOP.info

I get some stern warnings about doing this as root and that I probably shouldn't proceed this way.  Does anyone have any advice on what I should be doing differently here?

Thanks in advance.

wd15, 2025-04-09:
The Nix instructions as published on the Adamantine website might be out of date. The latest Nix instructions are [here](https://github.com/adamantine-sim/adamantine/blob/master/NIX.md). Possibly try running against HEAD with

    nix shell github:adamantine-sim/adamantine

That should drop you into a shell where you can run `adamatine`.

dpagendam, 2025-04-11:
Thank you @wd15 that got it working for me :-)

closed 2025-04-21 by Rombur