adamantine-sim/adamantine #381: Modelling Mechanical properties for the Hourglass Example

dpagendam, 2025-04-11:
Hi,

I'm using the Hourglass example as a learning exercise for Adamantine and am interested in looking at the mechanical properties such as displacement.  I have edited the file HourGlass_AOP.info so that the physics parameter "mechanical" was set to true.  I've then run the simulation using:

mpirun -np 1 adamantine --input-file=HourGlass_AOP.info

The simulation creates a series of .vtu and .pvtu files and I think the file names correspond to the physical state of the system at different time stamps.

When I open up a selection of the .vtu files in Paraview, but it looks like all of the properties like stress, displacement and temperature are zero for the points in the mesh.  I'm wondering what I am doing something wrong and why all of the physical properties are zero?

Any help or suggestions is greatly appreciated.

stvdwtt, 2025-04-15:
Hi @dpagendam, my guess is that if the only change you made was to set "mechanical" to true, that the solve didn't do anything because you didn't set the material parameters for mechanics. (I think the thermal expansion coefficient defaults to zero if you don't specify it.)

For an example of a thermoelastic simulation, see: https://github.com/adamantine-sim/adamantine/blob/master/tests/data/thermoelastic_bare_plate.info

There the solid phase of material_0 has `lame_first_parameter`, `lame_second_parameter`, and `thermal_expansion_coef` as parameters. If you look in the README, you can also a few parameters for turning on plasticity.

A word of warning -- the thermomechanics (and thermoplasticity in particular) is in a pretty immature state. We're still in the process of verifying and validating our implementation. That's something we're hoping to take care of in the next 6 months or so.

dpagendam, 2025-05-06:
Thank you again for this very helpful answer!